foxglove/mcap #16: Comments on the file format spec

dunk-emesent, 2022-01-14:
Hi,

I've been reviewing the MCAP spec, and have some feedback which may be of interest.

Attachments
- Why is these no publishing/acquisition time for an attachment? Not all attachment formats will support this metadata natively, so it would be useful to record it here.

Channel Info
- It may be useful to have a schema version, or store a hash of the schema. This allows consumers to determine whether they are able to process the data described by the schema, without having to reason about the contents of the schema itself.

Encryption / Signing
- No support.
- It would be useful to support encryption and signing on a per chunk basis.

Robustness and Chunks
- Major downsides of unchunked files are the lack of message indexes, and integrity checking.
- Would it be possible to use an approach whereby un-compressed chunks are written, along with message indexes? This would have negligable impact on write performance, and depending on the implementation (special case of an uncompressed chunk, where messages are written straight to file), would maintain robustness in the case of a crash.

defunctzombie, 2022-01-14:
Thanks for the feedback! We'll respond to all the raised items as various folks from the team chime in.

Here's my thoughts on the Robustness and Chunks section:

> * Major downsides of unchunked files are the lack of message indexes, and integrity checking.
> * Would it be possible to use an approach whereby un-compressed chunks are written, along with message [...]

The spec notes the following for the compression field of chunks: `A zero-length string indicates no compression`.

Our intent is to allow for uncompressed chunks to serve the needs of those that would like indexes and integrity checking while avoiding the overhead of compression. This does create a tradeoff for robustness during a process crash even for uncompressed chunks. Since the chunk record (like all records) is proceeded by a length, the content of the chunk (even if uncompressed) could be kept in memory. Depending on your recording situation, it could be possible to write zero'd placeholder bytes for the chunk length, stream the uncompressed messages into the record, and then go back and write the resulting length for the chunk.

I believe the format is flexible enough to allow for such a writer to exist even with length prefixed records. One design goal for mcap was to allow for append only writing, but it isn't a requirement that your writer be append only if your have other goals around crash safety and want to ensure all message bytes are written out and are ok doing some seeks to fill in lengths.

It might be worth us calling out these various suggestions in an implementation notes section or document.

wkalt, 2022-01-14:
As noted, thanks for the input.

My own thoughts:
* I like the idea of a schema version and understand the usefulness of it. A reader may also benefit from it by caching schemas and avoiding use of the end of file index, which could help on remote files.
* I wonder if attachments should support some kind of "metadata" section to allow this kind of thing (publish time) to be recorded. In discussion our thinking was that in many cases where publish time is what you are after, it will/could be in the file metadata, or furthermore that the attachment may be something like a log file where there isn't a single useful "publish time" but instead the file spans many times. With that in mind the thinking was "record time" was the only thing an mcap recorder could reliably depend on. However, I also see the utility of having a publish time available when the attachment does not contain it.
* On encryption - we don't feel we currently have enough input to confidently integrate it into the spec. We're interested to hear how users would like to use encryption, and whether chunk level encryption like you mention would be sufficient, or whether channel level per-encryption would also be required. For chunk level encryption, would you expect to encrypt different chunks with different keys? Per-channel encryption is another thing we have discussed. There is a draft PR for channel level encryption here: https://github.com/foxglove/mcap/pull/9, though it's mainly intended for discussion rather than consideration for merge.

amacneil, 2022-01-14:
It might be good to break these topics into separate tickets so we can discuss them independently.

Re attachment timestamps - I still think it would be ok to just have a `timestamp` field (could default to current time), and leave it up to the application to choose how to use that field. Attachments aren't going to be automatically written, so presumably the application adding attachments will have some understanding of how they are going to be used in the future.

wkalt, 2022-01-22:
I have split the schema_version suggestion into this issue: https://github.com/foxglove/mcap/issues/52

defunctzombie, 2022-01-23:
I've split the attachment request into this issue: https://github.com/foxglove/mcap/issues/56

defunctzombie, 2022-01-23:
Closing this issue. I believe we've created separate issues for all of the feedback items. @dunk-emesent feel free to comment on those issues if you have more to add/consider or make any new issues with additional feedback as we evolve the spec.

closed 2022-01-23 by defunctzombie