foxglove/mcap #23: _encoding_ is underspecified

defunctzombie, 2022-01-15:
The _encoding_ field within [ChannelInfo](https://github.com/foxglove/mcap/tree/main/docs/specification#channel-info-op0x03) is a string type with a few examples: _ros1_, _protobuf_, _cbor_. If I select _protobuf_ as the encoding, what should I put for _schema_?, _schema_name_? Does selecting an encoding have any other restrictions?

I'd like to write mcap reader/writer libraries that inter-operate with other tools but without additional specifications for how to handle encodings can't be sure what to do.

Some useful encodings to specify:
* ros1
* protobuf
* json
* cbor

amacneil, 2022-01-15:
Noting what we previously discussed -

This should be specified in a separate document that is linked from the main spec (since it would not be versioned with the main spec - adding a new format won’t require a spec version bump).

defunctzombie, 2022-01-26:
Covered on my branch: https://github.com/foxglove/mcap/tree/roman/spec-format/docs/specification

closed 2022-01-26 by defunctzombie