foxglove/mcap #36: Byte prefixed individual records (like arrays) feels weird

defunctzombie, 2022-01-21:
Issue for discussion

Since we have byte length prefixing for the entire record - having byte length for array fields feels weird. On the writer side I need to write out the array and only then do I know the length of bytes for the array.

wkalt, 2022-01-21:
My thoughts:

* since the whole record is prefixed with a length, you already need to materialize the array before you can write it out. Computing the byte length at that point is no added cost.
* on the read side, byte length prefixing allows skipping/projection of entire fields without measuring the length of each element - which seems like a clear win to me.

IMO it seems like the byte length gives you more flexibility than the count.

defunctzombie, 2022-01-21:
This is captured elsewhere - we should be clear that _length_ refers to byte length.

closed 2022-01-21 by defunctzombie