foxglove/mcap #40: Compression fields in both in Chunk Index and Chunk

defunctzombie, 2022-01-21:
Which of these is the authoritative one?

wkalt, 2022-01-21:
related thought discussed in slack: should the compression be in a new "chunk_stats" or similarly-named field in statistics, instead of its current location in the chunk index record where it is more like a covering index or included column (in pg for instance).

Arguments in favor:
* the compression info is not required for the use of the index that is reading chunk data. It's only needed as an included field for the "summary" operation.

Arguments against:
* the compression field is not the only field required by the summary operation, so either the new chunk_stats field needs to further duplicate information in the preceding chunk index records, or the chunk_stats field will not be comprehensive enough to serve the needs of the summary operation, which will then need to depend on both sources of data.

defunctzombie, 2022-01-21:
We will leave these fields and document that these fields are the non-authoritative versions.

defunctzombie, 2022-01-21:
Add message count field.

defunctzombie, 2022-01-26:
Captured in by branch: https://github.com/foxglove/mcap/tree/roman/spec-format/docs/specification

closed 2022-01-26 by defunctzombie