foxglove/mcap #48: How are readers supposed to determine that unchunked files contain messages?

jhurliman, 2022-01-21:
Here is a simple chunked file layout:

```
Chunked
-------
Magic
Header
Chunk
  ChannelInfo
  Message
MessageIndex
[index_offset]
ChannelInfo
ChunkIndex
Statistics
Footer
Magic
```

And the same file unchunked:

```
Unchunked
---------
Magic
Header
ChannelInfo
Message
[index_offset]
ChannelInfo
Statistics
Footer
Magic
```

And here's an empty file:

```
Empty (Chunked or Unchunked)
-------
Magic
Header
[index_offset]
Statistics
Footer
Magic
```

How are readers supposed to distinguish between unchunked files and empty files?

defunctzombie, 2022-01-21:
Also applies to chunked files that don't have indexes? If something doesn't have an index you go to beginning.

It can have an index data section but not message indexes (chunk indexes)

defunctzombie, 2022-01-21:
appendix section rather than the index section?

defunctzombie, 2022-01-21:
They read the index data section and see there are no chunk index records. Then they read from the top of the file.

If we update the spec to have statistics appear earlier - a bail-out would be reading the statistics record and seeing the chunk count === 0

defunctzombie, 2022-01-26:
https://github.com/foxglove/mcap/tree/roman/spec-format/docs/specification#footer-op0x7f

There are several options from the above working iteration:

If a footer has zero value for chunk_index_start then there are no chunks in the file and you can read from the start of the file to find messages.

If a footer has a non-zero value for statistics_start, you can read the statistics record and see if there are any messages.

If the footer has a non-zero channel_info_start, then there are at least channel info messages which _likely_ means there are messages to find in the file.

closed 2022-01-26 by defunctzombie