foxglove/mcap #52: include schema_version in channel info

wkalt, 2022-01-22:
A schema version field should be included in channel info. This could be a fixed-width 16 byte field (md5). This field can be used as a cache key by readers, to either avoid parsing (local) or downloading (remote) schemas they have observed in previous requests.

jhurliman, 2022-01-22:
I don’t think any newly drafted spec should be encouraging the use of md5.

For the local case I think local hashing as a reader optimization is something that could be done at read time without adding more to mcap files. Can you explain the remote case, and how a remote fetch (I assume range request on an mcap) could be avoided?

wkalt, 2022-01-24:
@jhurliman it doesn't have to be md5 and could be any width we like. It's not a security context, but avoiding md5 is probably wise for e.g fips compliance.

For the local case, the question here is "what should be used as the hash key". We don't currently provide a good option for that. You would need to hash the message definition before the lookup in the hash, which is work the feature in mcap avoids -- so at least by my understanding, this is not something you can do without a version field.

The situation I have used the equivalent field in ROS bags for is where I have a local reader with all the schemas for my organization in memory already, and remote files that all have the same set of schemas, and I want to start parsing at a known offset in the middle of the file (and read as fast as possible).

If I can quickly tell what schemas the file will contain, I can know whether I can start parsing from my offset directly or if I will need to separately download the schemas that will allow me to parse that data. The remote fetch being avoided is either from the end of the mcap files for those schemas, or from my database where I might also have them stored.

This can be implemented without a change to the format (by just synthesizing a "schema_version") from the hash of the schema, and returning it in all your API results for use as a cache key - but I can see the value of this being pretty broadly applicable to allow both customer orgs and backend tooling to make the same assumptions about the hash that should go there - otherwise orgs could have internal versions that mismatch their tooling's version for various reasons.

defunctzombie, 2022-01-26:
Added to my branch: https://github.com/foxglove/mcap/tree/roman/spec-format/docs/specification#channel-info-op0x03

closed 2022-01-26 by defunctzombie