foxglove/mcap #139: Add a way to indicate a schema name but no schema.

defunctzombie, 2022-02-09:
Let's say I have CBOR messages on channels. I'd like to give each "type" of message a "name" so that Studio can render my data in various panels. I don't want to (or need to) provide a schema since CBOR is self describing. What do I do?

defunctzombie, 2022-02-09:
Idea: the well-known encoding `CBOR` would specify allowing encoding as `""` (empty string) to indicate no `data` in the schema.

defunctzombie, 2022-02-09:
Update the language on the Schema record `encoding` field to indicate that an empty encoding string `""` means there is no schema data.

amacneil, 2022-02-10:
If there is no `Schema.data` it seems that would indicate there is no schema, regardless of the claimed `Schema.encoding` right?

defunctzombie, 2022-02-10:
We wanted something more explicit for spec readers to consult that makes it clear schema doesn't need to be present and to do that you don't need to specify an encoding.

PR #147 (2022-02-10): spec: Allow empty encoding string in Schema record
Fixes: #139
commit 19477265 (2022-02-10): spec: Allow empty encoding string in Schema record (#147)

Fixes: #139