foxglove/mcap #184: Go: Segmentation violation converting .db3 to .mcap

bertaveira, 2022-02-12:
**Description**
I am trying to convert .db3 files to .mcap using the go command line tool but I am getting an error. I have tried both on MacOS and Linux (both ARM64)

Error I get:
```
./mcap convert multiple_files_0.db3 multi.mcap --directories /opt/ros/foxy/
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x695880]

goroutine 1 [running]:
github.com/foxglove/mcap/go/ros.collectMessageSchemas.func1({0x4000024012, 0x3e8}, {0x0, 0x0}, {0x915fa0, 0x40001c8180})
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/ros/ros2db3_to_mcap.go:30 +0x30
path/filepath.Walk({0x4000024012, 0x3e8}, 0x40001937b8)
	/usr/local/go/src/path/filepath/path.go:503 +0x68
github.com/foxglove/mcap/go/ros.collectMessageSchemas({0x4000140a80, 0x2, 0x2}, {0x4000119b10, 0x1, 0x1})
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/ros/ros2db3_to_mcap.go:29 +0x17c
github.com/foxglove/mcap/go/ros.DB3ToMCAP({0x916120, 0x4000134120}, 0x400011de10, 0x4000193d48, {0x4000140a80, 0x2, 0x2})
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/ros/ros2db3_to_mcap.go:160 +0x218
github.com/foxglove/mcap/go/mcap/cmd.glob..func2(0xb5e320, {0x400015c880, 0x2, 0x4})
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/mcap/cmd/convert.go:119 +0x4f8
github.com/spf13/cobra.(*Command).execute(0xb5e320, {0x400015c840, 0x4, 0x4})
	/home/bertaveira/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:860 +0x650
github.com/spf13/cobra.(*Command).ExecuteC(0xb5eaa0)
	/home/bertaveira/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:974 +0x424
github.com/spf13/cobra.(*Command).Execute(...)
	/home/bertaveira/go/pkg/mod/github.com/spf13/cobra@v1.3.0/command.go:902
github.com/foxglove/mcap/go/mcap/cmd.Execute()
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/mcap/cmd/root.go:19 +0x30
main.main()
	/media/psf/Home/Documents/CFS/dv/random/mcap/go/mcap/main.go:21 +0x20
```

- Version: main branch
- Platform: MacOS and Linux (both ARM64)

**Steps To Reproduce**

I have tried with both a rosbag I have and a sample one supplied for testing and I get the same error. Rosbag here: https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/resources/reindex_test_bags/multiple_files/multiple_files_0.db3

I then compile using `make build` and then try to convert it like so:
```
./mcap convert multiple_files_0.db3 multi.mcap --directories /opt/ros/foxy/
```

wkalt, 2022-02-12:
@bertaveira Thanks for the report. I wasn't able to reproduce this on a stock install of ros2 foxy (although I am on x64 -- I doubt this is architecture related), but I did just merge a fix for the bug that is causing this to panic, rather than report an error. Would you mind pulling down the latest and reporting what you see now?

bertaveira, 2022-02-12:
Yes the Panic is fixed. However I now get an error related to message types that are not a part of the bag file I am trying to convert. 

E.g: converting the sample bag linked above I get these errors:
```
./bin/mcap convert multiple_files_0.db3 multi-test.mcap
failed to convert file: failed to list provided directory /Users/bertaveira/ros2_foxy/install:/Users/bertaveira/Documents/CFS/dv/ROS_packages/cfs_msgs/install/cfs_msgs: lstat /Users/bertaveira/ros2_foxy/install:/Users/bertaveira/Documents/CFS/dv/ROS_packages/cfs_msgs/install/cfs_msgs: no such file or directory
```

wkalt, 2022-02-13:
Thanks. From your error I realized that the environment variable AMENT_PREFIX_PATH is holding a colon-separated list of paths, rather than a single filepath. I have just merged a fix to split the prefix path so all directories are searched. Can you try again?

bertaveira, 2022-02-13:
Glad I could help. Since the last commit (d778d34) I can't build it. I get the same error in linux and MacOS:
```
$ make build
mkdir -p bin
go build -o bin/mcap -tags sqlite_omit_load_extension,netgo,osusergo -ldflags '-s -w -extldflags "-static"' .
# github.com/foxglove/mcap/go/cli/mcap/cmd
cmd/convert.go:116:7: undefined: prefix
make: *** [Makefile:13: build] Error 2
```

wkalt, 2022-02-14:
Hi, very sorry about that. Please try again now.

bertaveira, 2022-02-14:
Yes it seems to have been fixed. It both compiles and is able to convert. 

The file size is a bit suspicious since it is a 1/10 of the db3 file. It also complains about not finding some schemas that are in the directory supplied. Running multiple times in a row the error of missing schemas changes until it fully disappears. 

I would close this issue since I have no way to test it for now. I will wait until the new release of foxglove with support for mcap arrives to test it further since the current one gives an error of initialisation opening the mcap files. Once I can test I'll report if anything comes up. Thanks for the quick fixes!

closed 2022-02-14 by bertaveira

wkalt, 2022-02-14:
@bertaveira ok, interesting. That error about missing schemas sounds like a clear bug - I could maybe imagine different failures per run if the iteration over definitions were nondeterministic, but for it to go away sounds broken. I'll file an issue for that and investigate separately.

One thing you can do to test the contents of the file, is try running `mcap info` or `mcap cat` on it. The latter will print a line per message in the file, with the topic next to it. It won't parse message contents yet.

Thanks for trying the tool out!