foxglove/mcap #255: go: writer will record empty chunk index when only a schema or channel occurs in the chunk

wkalt, 2022-02-28:
When a chunk contains only a schema or channel, but no messages, the writer will still record a chunk index record to the end of the file on close. However, this index will have a start time of zero, which is incorrect, but indistinguishable from a chunk containing a record legitimately dated zero.

The writer needs to avoid writing chunk indexes for such chunks.

amacneil, 2022-03-01:
Do we have conformance tests or something to cover this across implementations?

jtbandes, 2022-03-02:
Why is it important to distinguish empty chunks from chunks with messages at time zero?

wkalt, 2022-03-04:
@jtbandes important is subjective but IMO,
* better to avoid ambiguous possibilities all things equal
* current situation means readers need to make an extra seek/range request for chunks that are actually empty
* current situation means there is no reliable way for a reader to determine the start time of a file, if the statistics record is not recorded but the chunk indexes are

james-rms, 2022-09-14:
Changing this to a "feature" ticket, because it's hard to point to a tool or user experience that is degraded due to this issue.

Re: optional message count field - at the moment we don't have a clear definition for what an "optional" field is. If we add a new field to the end of ChunkIndex, this might be OK - old parsers would just treat it as padding. However, if an old writer is inserting padding for its own reasons, that might be interpreted as an invalid message count. 

I'm not sure what the deal is with padding, I know we have conformance tests about it but I don't know why we'd insert it otherwise (alignment?).