foxglove/mcap #259: Data discontinuities in data coming from data platform are not plotted correctly.

foxymiles, 2022-03-01:
**Description**
Data discontinuities should be skipped in plots but currently we connect them in data streamed from data platform.

- Version: 1.2.0
- Platform: All

**Steps To Reproduce**
1. Open a long file from data platform.
2. Plot some topic from the file.
3. Use the play bar to jump to different sections of the file.
4. Observe the plot try to connect the gaps.
[image]

**Expected Behavior**
Gaps should be excluded from the plot.

**Actual Behavior**
Gaps are connected.

wkalt, 2022-03-02:
@foxymiles is this an "mcap" issue or a studio one?

jtbandes, 2022-03-02:
Studio

foxymiles, 2022-03-03:
Closing in favor of https://github.com/foxglove/studio/issues/2956

closed 2022-03-03 by foxymiles