foxglove/mcap #287: Store an index by publish time

defunctzombie, 2022-03-11:
The current index records are great when you want to read the messages by log_time. If you want to read the messages by publish_time, you have to create your own index by reading all the records and building it in memory. It would be nice to consider writer options for recording both index types in the file.

wkalt, 2022-03-17:
I think the best solution would be to store (log time, publish time, offset) triples in the message index record and extend the chunk index record with publish start/end. Breaking change for the former + 8 bytes per record.

If we need something in the interim, we could append publish_start and publish_end to the chunk index record as optional fields, and append a publish_time_index array to the end of message index that mirrors the existing record time index structure. This would cost 8 bytes per record more than the breaking solution due to timestamp duplication.

shuhaowu, 2025-11-07:
Can the concept of time be dynamically changed based on a flag somewhere? Maybe in the footer there could be a flag that dictates if "time" means `log_time` or `publish_time`? A binary compatible way would be to throw that information in the header.library, but that would require a read of the header which is an extra required read.

There are a few places where time means `log_time` for now:

- In ChunkIndex, the `message_start_time` and `message_end_time` is currently `log_time`.
- In MessageIndex, the `records` has its values as `log_time`.
- In Chunk, the `message_start_time` and `message_end_time` is currently `log_time`.

If a flag can universally switch them to mean `publish_time`, then no breaking change would be needed. Unfortunately I don't see any flexible/reserved fields for this other than the `Header.profile` or `Header.library`. I suppose could also use `Metadata` but that would require possibly up to 2 additional IO (at least 1 additional) which can be cost prohibitive in high-performance scenarios.

shuhaowu, 2025-11-07:
The alternative is to maybe introduce another record type with a new opcode: ChunkIndex2 and MessageIndex2 which has this information. The `2` version would be mutually exclusive with the original version.

shuhaowu, 2025-11-08:
Yeah, having more than 2 timestamps is also a very important use case (more below). But really having the publish time as index is more of a show stopper. Having a `2` version that gracefully degrade would solve that problem.

It is actually possible to support per message timestamps and N timestamps without breaking the binary compatibility. The way to do this is to create a new payload type that itself can encapsulate protons/flat buffer/JSON/etc. Basically the message encoding for a channel would be something like `encapsulated_proto` or `encapsulated_json`. Then the data could contain the metadata and the timestamps as a prefix, and the proto and JSON can follow it. It's not the prettiest solution, but it will work. So like:

Message:

- publish_time
- data
  - timestamps: uint64[]
  - metadata: ...
  - inner_data: byte

In a way this is the same as if you changed the binary protocol (imagine if the above is flattened), except the versioning is controlledb y the channel message_encoding field, if that makes sense.

shuhaowu, 2025-11-08:
If there is interest in this, I may have some time to create a proposal/PR.