foxglove/mcap #336: Does not work with bags containing hidden topics

jmarsik, 2022-03-25:
**Description**

- Version: mcap-cli 0.0.4
- Platform: Linux AMD64

**Steps To Reproduce**

Record ROS2 bag file with `--include-hidden-topics` option. Perform some action server call while recording.
Then try to convert this bag file to MCAP format with `mcap convert filename.db3 filename.mcap`.

**Expected Behavior**

File converts successfully and the resulting MCAP file will contain schema definitions even for the hidden topics.

**Actual Behavior**

File will not convert and an error will be displayed:

```
failed to convert file: failed to find schema for control_msgs/action/FollowJointTrajectory_FeedbackMessage: schema not found
```

This example is for a bag file containing data from calling the FollowJointTrajectory action.

wkalt, 2022-03-25:
@jmarsik thanks for the report. I was able to reproduce the issue.

The root problem is that the conversion tool today does not support actions, only messages. The hidden topics are actions, at least in this case. This isn't by design, it was just an oversight in the ros2 support.

To support actions properly, we will first need to extend the ros2 mcap profile here: https://github.com/foxglove/mcap/blob/main/docs/specification/profiles/ros2.md to describe how they should be serialized in mcap. I raised https://github.com/foxglove/mcap/issues/336 to cover that work.

For this issue I am going to focus on ensuring that actions are ignored by the tool instead of causing this error.

wkalt, 2022-03-28:
@jmarsik I've implemented the behavior above (skipping actions) in v0.0.5, which is available for download here: https://github.com/foxglove/mcap/releases/download/releases%2Fmcap-cli%2Fv0.0.5/mcap-linux-amd64.