foxglove/mcap #337: Support ros2 actions

wkalt, 2022-03-25:
Currently the mcap ros2 profile does not specify how actions should be recorded. The profile should be updated to cover actions, and the CLI tool's ros2 bag converter also updated with the desired approach.

wkalt, 2022-03-27:
I recorded a bag with some actions using the turtlesim node and my arrow keys for navigation. Observations:
* The type of the action topic in the topics table is `turtlesim/action/RotateAbsolute_FeedbackMessage`.
* Unlike with message topics, there is no corresponding file under the ament prefix path in `share/turtlesim/action/RotateAbsolute_FeedbackMessage.{action, idl}`. Instead, the related files are `share/turtlesim/action/RotateAbsolute.{action, idl}`. The contents are,

turtlesim/action/RotateAbsolute.idl

```
// generated from rosidl_adapter/resource/action.idl.em
// with input from turtlesim/action/RotateAbsolute.action
// generated code does not contain a copyright notice

module turtlesim {
  module action {
    @verbatim (language="comment", text=
      " The desired heading in radians")
    struct RotateAbsolute_Goal {
      float theta;
    };
    struct RotateAbsolute_Result {
      @verbatim (language="comment", text=
        " The angular displacement in radians to the starting position")
      float delta;
    };
    struct RotateAbsolute_Feedback {
      @verbatim (language="comment", text=
        " The remaining rotation in radians")
      float remaining;
    };
  };
};

```

RotateAbsolute.action:
```
# The desired heading in radians
float32 theta
---
# The angular displacement in radians to the starting position
float32 delta
---
# The remaining rotation in radians
float32 remaining
```
* From the contents, it seems like the IDL file is the only viable one to put in the mcap file, since the `.action` file contains no way to make sense of the term `FeedbackMessage`.

Compounding the confusion, the messages on this topic are not actually recorded by ros2 record - just the topics record. So it isn't easy to check the data and understand what is necessary to parse it.

jmarsik, 2022-03-28:
If you need message definitions for them (.msg files), you can split the .action file by `--` lines and artificially create 3 .msg files with corresponding suffixes.

But it's a surprise to me that `ros2 bag record` does not record them, I thought that it does! If you create those artificial .msg definitions, would it be possible to use your storage driver to record them?

Aposhian, 2022-04-14:
>compounding the confusion, the messages on this topic are not actually recorded by ros2 record - just the topics record. So it isn't easy to check the data and understand what is necessary to parse it.

`ros2 bag record` can record these, but they are hidden topics (denoted by the `_action` namespace), which are not recorded by default. You can record them by either adding `--include-hidden-topics` or by explicitly listing them as a topic to record like
```
ros2 bag record /my_action/_action/feedback /my_action/_action/status`
```