foxglove/mcap #354: go "make test" doesn't pass

jameskuszmaul-brt, 2022-03-30:
**Description**

No clue if this is even meant to be supported, since the CI seems to run the tests in `go/conformance` indirectly through some typescript scripts, but since there are Makefiles with test targets in `go/`, I would've expected `cd go/conformance; make test` to work, but instead the tests appear to fail.

jtbandes, 2022-03-30:
Can you share what errors you got? I just tried this:

```
$ make test
make -C test-streamed-write-conformance test
go test ./...
ok      github.com/foxglove/mcap/go/conformance/test-streamed-write-performance 0.154s
make -C test-streamed-read-conformance test
go test ./...
ok      github.com/foxglove/mcap/go/conformance/test-streamed-read-conformance  0.186s
```

jameskuszmaul-brt, 2022-03-30:
Oh it's `git-lfs` nonsense. Still not used to having repos that use `git-lfs`, so I'm not used to pulling that first....

closed 2022-03-30 by jameskuszmaul-brt