foxglove/mcap #370: x- prefix is recommended for Schemas, at odds with RFC 6648

jhurliman, 2022-04-11:
There is no justification given for the x- prefix recommendation. The only historical origin I can find for this convention is from HTTP, which deprecated the practice in 2012.

https://www.rfc-editor.org/rfc/rfc6648

closed 2022-07-12 by jhurliman

alkasm, 2022-08-07:
The `x-` prefix is required in code by `mcap doctor`: https://github.com/foxglove/mcap/blob/7f8ac8f11c9ea78a057468379ad599da18d01f8a/go/cli/mcap/cmd/doctor.go#L239-L243

And still recommended in Python comments:
* https://github.com/foxglove/mcap/blob/a254fa07a7680856fdaf6136190dc8970c2045bd/python/mcap/mcap/mcap0/well_known.py#L5-L6
* https://github.com/foxglove/mcap/blob/2f51f478e16f219387df192b86925f760230e90e/python/mcap/mcap/mcap0/writer.py#L340-L342
* https://github.com/foxglove/mcap/blob/2f51f478e16f219387df192b86925f760230e90e/python/mcap/mcap/mcap0/writer.py#L367-L368

jtbandes, 2022-08-08:
https://github.com/foxglove/mcap/pull/521