foxglove/mcap #374: Ability to automatically include common ros 2 message defs

amacneil, 2022-04-27:
Let me convert db3 files using common ros message defs, without needing them on my filesystem

https://github.com/ros2/common_interfaces

jhurliman, 2022-09-22:
Now that Humble is released with at least one breaking message definition change (`Marker`), supporting common ROS2 definitions is no longer as simple as bundling an arbitrary snapshot of `common_interfaces` and `rcl_interfaces`. I propose we close this ticket.

amacneil, 2022-09-22:
Is it harder than having both options available, and during `mcap convert` attempt to decode on a single Marker message on that topic to identify the correct version to add to the output mcap file?

jhurliman, 2022-09-22:
@amacneil there is no CRC/hashing or any feedback at decode time that your decoding is the correct interpretation of the serialized data. You could try both, and if one decoder goes past the end of the serialized data and the other doesn't then assume one is correct. But it's possible to have a serialized message that decodes with either message definition, but one would produce nonsense values.

amacneil, 2022-09-22:
>  it's possible to have a serialized message that decodes with either message definition, but one would produce nonsense values

😭

closed 2022-09-22 by amacneil