foxglove/mcap #375: Ability to load ros 2 message files directly from filesystem

amacneil, 2022-04-27:
Rather than needing `--ament-prefix-path`, let me specify a folder to recursively search for the appropriate missing msg files.

V0XNIHILI, 2022-08-27:
Do you know how to point to the missing schemas? Pointing to my ROS2 project folder does not seem to help.

jtbandes, 2022-08-29:
@V0XNIHILI  Have you tried using `--ament-prefix-path` as described here? https://mcap.dev/getting-started/ros-2

V0XNIHILI, 2022-08-30:
Yes, I have tried that. When I pass the contents of the environment variable `AMENT_PREFIX_PATH` with that flag, the same happens. However, the error comes back so fast after pressing enter that I have the idea that it does not really consider my argument.

I run it like this:

```bash
./mcap convert rosbag2_2022_08_09-14_19_47_0.db3 test.mcap --ament-prefix-path /home/parallels/Projects/ROS2-Project/install/viz:/home/parallels/Projects/ROS2-Project/install/testing:/home/parallels/Projects/ROS2-Project/install/simulator:/home/parallels/Projects/ROS2-Project/install/fake_state_estimation:/home/parallels/Projects/ROS2-Project/install/fake_sensors:/home/parallels/Projects/ROS2-Project/install/fake_perception:/home/parallels/Projects/ROS2-Project/install/ava_state_estimation:/home/parallels/Projects/ROS2-Project/install/vehicle_config:/home/parallels/Projects/ROS2-Project/install/rviz_controller_plugin:/home/parallels/Projects/ROS2-Project/install/fake_embedded:/home/parallels/Projects/ROS2-Project/install/fake_dynamics:/home/parallels/Projects/ROS2-Project/install/controls_bypass:/home/parallels/Projects/ROS2-Project/install/collision_detector:/home/parallels/Projects/ROS2-Project/install/ava_planning:/home/parallels/Projects/ROS2-Project/install/sim_msgs:/home/parallels/Projects/ROS2-Project/install/se_msgs:/home/parallels/Projects/ROS2-Project/install/rviz_interface_plugin:/home/parallels/Projects/ROS2-Project/install/rosbag_frame_loader:/home/parallels/Projects/ROS2-Project/install/mission_control:/home/parallels/Projects/ROS2-Project/install/labeling_assist:/home/parallels/Projects/ROS2-Project/install/labeling_msgs:/home/parallels/Projects/ROS2-Project/install/keyboard_controls:/home/parallels/Projects/ROS2-Project/install/cms_car:/home/parallels/Projects/ROS2-Project/install/can_interface:/home/parallels/Projects/ROS2-Project/install/ava_fake_trackdrive:/home/parallels/Projects/ROS2-Project/install/embedded_msgs:/home/parallels/Projects/ROS2-Project/install/ava_perception:/home/parallels/Projects/ROS2-Project/install/cv_msgs:/home/parallels/Projects/ROS2-Project/install/cms_car_msgs:/opt/ros/galactic
```

Then, the output is:

```bash
failed to convert file: failed to find schema for embedded_msgs/msg/ASControlsEstimations: schema not found
```

vj2181, 2022-09-07:
Did you figure this out - @V0XNIHILI ? I have the same error - not sure how to point the tool to the correct message definition.

V0XNIHILI, 2022-09-07:
No, unfortunately not :(

Maybe @jtbandes, do you have some ideas/suggestions? Because I am unable to use Foxglove Studio without the MCAP CLI tool working

vj2181, 2022-09-07:
In my case, echo $AMENT_PREFIX_PATH already has the correct directories with the message definitions - still the CLI tool fails with "schema not found" message. Explicitly specifying the directories does not help as well. Will wait and see if anyone else replies here.

V0XNIHILI, 2022-09-22:
@jtbandes would you have time to take a look at this? Because this leaves us unable to use the Foxglove Data Platform

amacneil, 2022-09-22:
@james-rms putting on your radar

jtbandes, 2022-09-22:
@V0XNIHILI @vj2181 are you folks using .msg files or .idl?

V0XNIHILI, 2022-09-22:
I am using `.msg` files

V0XNIHILI, 2022-10-03:
Any update on this?

amacneil, 2022-10-05:
If you are having difficulty with `mcap convert`, I recommend trying `ros2 bag convert` instead. It should achieve the same thing but will be more reliable at finding message definitions since it has access to the ros workspace.

james-rms, 2022-11-01:
Closing this issue as not planned - for complex workspaces I recommend using `ros2 bag convert` with the mcap storage plugin.

closed 2022-11-01 by james-rms