foxglove/mcap #376: Improve error message when there is a missing schema

amacneil, 2022-04-27:
When running mcap convert, if there is a missing message schema, I get:

```
$ ~/src/mcap/go/cli/mcap/bin/mcap convert test.db3 test.mcap
failed to convert file: failed to find schema for sensor_msgs/msg/PointCloud2: schema not found
```

This error should tell me what to do next to fix the issue. Ideally related to https://github.com/foxglove/mcap/issues/375 so I can just point it at the missing message file.

martijnhabers, 2024-01-08:
Just a heads up that its my first time commenting on an issue so if things are out of place, that's why and please point me in the right direction :)

I'm facing a similar problem. I saw that issue #375 was closed, where it was recommended to use `ros2 bag convert` instead. However, I am not always on a system with a ROS2 installation (currently on an M2 Mac),  which is why I'm interested in using the mcap cli tool to convert db3 to mcap files with custom messages.

I've tried using the `--ament-prefix-path` argument, with no successful result of finding custom messages. I have a list of defined custom messages (in .msg format) but can not find a way to convert to mcap without sourcing a ROS2 build.

I can't find any changes since these issue that makes it possible/easier to point to message definitions, do you have any update or idea on if this is possible?

wkalt, 2024-01-08:
@martijnhabers thanks for reporting. We were actually having an internal discussion last week about making this better, so this is timely.

First, hopefully some advice to get it working. The  `--ament-prefix-path` argument is expecting a directory structure that exactly matches what ROS puts in e.g `/opt/ros/galactic`. To get the tool working I would look at the test data examples here: https://github.com/foxglove/mcap/tree/main/go/ros/testdata/get_schema_workspace/share, and look at how the structure of the index file in `ament_index/resource_index/rosidl_interfaces` corresponds to the messages under `example_messages`. If you build a similar directory tree, rooted under "share", and stick "share" in the directory your `--ament-prefix-path` points at, I think you should have success (unless we have other issues).

Second, what we're discussing internally is how we can make the schema detection behavior smarter, including whether we need to mandate a particular directory layout or whether there is a common one users are ending up with that we should support. Would you be able to share a sanitized example of the layout of the directory you are working with?

martijnhabers, 2024-02-16:
Hi, thanks for the reply, haven't really been working on it this past month but now I want to see if I can get it working.

A simplified structure we have implemented for custom messages used in our workspace looks like this:

```
└── 📁ros2
    └── 📁src
        └── 📁custom_ros_package
        └── 📁custom_ros_package_2
        └── 📁shared
            └── 📁shared_msgs
                └── 📁msg
                    └── one.msg
                    └── two.msg
                    └── etc.msg
                └── 📁srv
                    └── one.srv
                    └── two.srv
                    └── etc.srv
```

I have tried to use what you mentioned and mimic the directory structure in opt/ros/foxy 

```
└── 📁ros2
    └── 📁src
        └── 📁custom_ros_package
        └── 📁custom_ros_package_2
        └── 📁share
            └── 📁ament_index
                └── 📁resource-index
                    └── 📁rosidl_interfaces
                        └── shared_msgs
            └── 📁shared_msgs
                └── 📁msg
                    └── one.msg
                    └── two.msg
                    └── etc.msg

```

However, when running the command `mcap convert input_file.db3 output_file.mcap --ament-prefix-path /ros2/src`it should be able to parse the share folder for the message schemas right? However, it still returns the same "failed to find schema".

Hope this clears things up a bit.