foxglove/mcap #380: CLI tool reports 0.00 Hz for all topics

jhurliman, 2022-05-05:
```
  	(1) /rosout                                       0 msgs (0.00 Hz)   : rcl_interfaces/msg/Log [ros2msg]
  	(2) /parameter_events                             0 msgs (0.00 Hz)   : rcl_interfaces/msg/ParameterEvent [ros2msg]
  	(3) /camera/aligned_depth_to_color/camera_info  129 msgs (0.00 Hz)   : sensor_msgs/msg/CameraInfo [ros2msg]
  	(4) /camera/aligned_depth_to_color/image_raw    119 msgs (0.00 Hz)   : sensor_msgs/msg/Image [ros2msg]
  	(5) /camera/color/camera_info                   123 msgs (0.00 Hz)   : sensor_msgs/msg/CameraInfo [ros2msg]
  	(6) /camera/color/image_raw                     116 msgs (0.00 Hz)   : sensor_msgs/msg/Image [ros2msg]
  	(7) /camera/depth/color/points                  123 msgs (0.00 Hz)   : sensor_msgs/msg/PointCloud2 [ros2msg]
  	(8) /camera/depth/camera_info                   125 msgs (0.00 Hz)   : sensor_msgs/msg/CameraInfo [ros2msg]
  	(9) /camera/depth/image_rect_raw                124 msgs (0.00 Hz)   : sensor_msgs/msg/Image [ros2msg]
  	(11) /tf                                          0 msgs (0.00 Hz)   : tf2_msgs/msg/TFMessage [ros2msg]
  	(12) /tf_static                                   1 msgs (0.00 Hz)   : tf2_msgs/msg/TFMessage [ros2msg]
```

jtbandes, 2022-05-05:
I probably broke this somehow in #369

jhurliman, 2022-05-05:
> we should add some testing for the output formatting

Maybe this should be its own ticket

wkalt, 2022-05-06:
Original bug fixed in https://github.com/foxglove/mcap/pull/381. Ticket for testing here: https://github.com/foxglove/mcap/issues/382.

closed 2022-05-06 by wkalt