foxglove/mcap #389: reading converted ros1 to mcap files, gives out an error in python libary

Dragonking3x, 2022-05-12:
**Description**
After converting a ros1 *.bag file to *.mcap with
**`mcap convert rosbag.bag rosbag.mcap`**
the python library can't read more than the first record.

First record:
`Header(profile='ros1', library='mcap go #(devel)')`
ERROR:
`zstd.Error: Input data invalid or missing content size in frame header.`

Converting it with
**`mcap convert --chunked=False rosbag.bag rosbag.mcap`**
makes it able to be used with the python libary.

**Version**
mcap-cli: mcap-linux-amd64 (v0.0.8)
Language: Python 3.8.10
libary: mcap (0.0.8)
Platform: Ubuntu 20.04

**Steps To Reproduce**
- Use a ros1 *.bag file.
- mcap convert rosbag.bag rosbag.mcap
[image]

**Behavior**
/home/dragonking3x/PycharmProjects/testing/venv/bin/python /home/dragonking3x/PycharmProjects/testing/main.py
Header(profile='ros1', library='mcap go #(devel)')
Traceback (most recent call last):
  File "/home/dragonking3x/PycharmProjects/testing/main.py", line 100, in <module>
    main('PyCharm')
  File "/home/dragonking3x/PycharmProjects/testing/main.py", line 28, in main
    read_mcap(file_name)
  File "/home/dragonking3x/PycharmProjects/testing/main.py", line 34, in read_mcap
    for record in reader.records:
  File "/home/dragonking3x/PycharmProjects/testing/venv/lib/python3.8/site-packages/mcap/mcap0/stream_reader.py", line 91, in records
    chunk_records = breakup_chunk(record)
  File "/home/dragonking3x/PycharmProjects/testing/venv/lib/python3.8/site-packages/mcap/mcap0/stream_reader.py", line 31, in breakup_chunk
    stream, stream_length = get_chunk_data_stream(chunk)
  File "/home/dragonking3x/PycharmProjects/testing/venv/lib/python3.8/site-packages/mcap/mcap0/stream_reader.py", line 53, in get_chunk_data_stream
    data: bytes = ZSTD_uncompress(chunk.data)
zstd.Error: Input data invalid or missing content size in frame header.
INFO - 2022-05-12 12:14:05,747 - core - signal_shutdown [atexit]

jtbandes, 2022-05-12:
Thanks for reporting. I can reproduce the issue. This is a bug in the Python library.

jtbandes, 2022-05-12:
Fixed in 0.0.9

PR #390 (2022-05-12): python: fix zstd decompression
**Public-Facing Changes**
Fixed zstd decompression in the Python reader.

**Description**
Fixes #389
The `zstd` package did not allow us to pass the uncompressed size as a parameter to the decompress function. The `zstandard` package does. This is required when the writer used a streaming compression API.

Tests are not added because the Python writer uses the one-shot zstd compress API, so does not trigger this issue in the reader (which only occurs when the compressed frame does not include a size value, i.e. when a streaming compression API was used).
commit 5dd62527 (2022-05-12): python: fix zstd decompression (#390)

**Public-Facing Changes**
Fixed zstd decompression in the Python reader.

**Description**
Fixes #389
The `zstd` package did not allow us to pass the uncompressed size as a parameter to the decompress function. The `zstandard` package does. This is required when the writer used a streaming compression API.

Tests are not added because the Python writer uses the one-shot zstd compress API, so does not trigger this issue in the reader (which only occurs when the compressed frame does not include a size value, i.e. when a streaming compression API was used).
closed 2022-05-12 by jtbandes