foxglove/mcap #393: Confusing install instructions on mcap.dev

jhurliman, 2022-05-16:
**Description**

The mcap.dev homepage lists two separate methods for installation, but doesn't provide any context for when to use one method or the other. What are the relative advantages of each approach?

jhurliman, 2022-05-16:
I don't think there is any benefit to the golang-based route. My understanding is that is only for golang developers that have a go compiler and related tooling installed on their machine, and want to build the CLI tool from source. It makes sense in the context of the GitHub repo next to the source code where you may want to check out and modify the source code, but I don't think it makes sense on a landing page for MCAP that is not assuming the reader is a go developer or wants to modify the source code of the MCAP cli tool.

esthersweon, 2022-05-16:
Got it, so this issue should be to remove the `go` command entirely from the mcap.dev landing page.

closed 2022-05-16 by esthersweon