foxglove/mcap #403: Add info on landing page about how you should record/convert ROS 2 files on the same distro to avoid issues

esthersweon, 2022-05-26:
**Description**
- Add info on landing page about how you should record/convert ROS 2 files on the same distro to avoid issues

esthersweon, 2022-08-15:
@amacneil @defunctzombie - this seems more like a callout about how foxglove CLI works than about how MCAP the file format does. 

I propose we call this out in the [CLI repo](https://github.com/foxglove/mcap/tree/main/go/cli/mcap), instead of the MCAP website like was originally intended when opening this ticket.

amacneil, 2022-08-16:
I agree it's a caveat of the CLI rather than MCAP itself. So we should note it in the CLI docs.

However, I think we should move the CLI docs out of that readme into our new docs site.