foxglove/mcap #438: CLI: mcap cat --csv --topics /tf

wkalt, 2022-06-23:
It would be nice if we could spit out a CSV file with column headers. If a user wanted to look at the history of a single topic in R or python this would make that very easy to do that.

I think it would probably require that the --topics flag be specified with only one topic. Then we could define a JSON->CSV translation and pipe the JSON output to it. Seems easier to work off the newline-delimited JSON output than write new ros/protobuf converters for CSV.

sisaha9, 2023-02-10:
Has there been any work started on this?

jtbandes, 2023-02-10:
Currently no work has been started. Could you describe your use case and desired functionality, to aid the discussion?

sisaha9, 2023-02-10:
We had a similar tool developed for SQL based ROS2 bags. We could pass a ROS2 bag, a topic and an output file name and we would get a CSV file with all the information on that topic. I like the CLI method mentioned here. Would love to contribute on it but not too familiar with Go. If there was a way to do this via the Python API I would like to take a look at it

jtbandes, 2023-02-11:
You could absolutely achieve this with a Python script. Here are some Python examples to get you started: https://mcap.dev/guides/python/ros2.html

sisaha9, 2023-02-11:
Do you have any suggestions on how to get the list of attributes for a message (including nested). A DFS could then be used to construct the CSV file

sisaha9, 2023-02-11:
Or if it's possible to convert to JSON via a Python method