foxglove/mcap #446: mcap fix command

james-rms, 2022-06-27:
We want a way for CLI users to fix mcaps that are incomplete or corrupted, but fixable. CLI is fluid at this point - could do one of:

```
mcap fix FILE [--output OUTFILE] [--in-place] [--destructive]
mcap doctor FILE [--fix] [--output OUTFILE] [--in-place] [--destructive]
```

amacneil, 2022-06-28:
Maybe rather than “fix” could use the term “recover”. The implication is that we are working with a corrupt file, so we will do our best to recover it and print warnings for anything unrecoverable.

defunctzombie, 2022-08-01:
James: the initial scope is a "reindex".
Jacob: read as much as you can and write out to a new file?
James: it will always write out to a new file.

defunctzombie, 2022-08-01:
Adrian: maybe we should call it "recover"

* Adrian is internally consistent over 1 month.

defunctzombie, 2022-08-01:
Wyatt: `mcap check --fix`

defunctzombie, 2022-08-01:
Roman: `mcap doctor --surgery`

defunctzombie, 2022-08-15:
We've dropped this from the mcap devx epic for now.

jhurliman, 2022-09-22:
When naming for this command, consider it can be used for the following use cases:

1. I have a valid MCAP file, but it was recorded with small chunk sizes for crash resiliency and I want to reencode it with larger chunk sizes for better compression.
2. I have a truncated MCAP file from a crashed robot. I want to copy all of the valid data out of this file to a new, properly indexed MCAP.
3. I have an MCAP with CRC failures on one or more chunks due to flaky storage media. I want to copy all the data that passes CRC validation into a new MCAP.
4. I have an MCAP with out-of-order messages. I want to create a copy where the log_time of messages is monotonic.
5. I have an uncompressed MCAP, I want to create a compressed version.

defunctzombie, 2022-09-22:
@jhurliman Good thoughts - after reading those my initial reaction is maybe it is worth having separate commands. One is meant for files that are "invalid" or "broken" in some way (with a set of flags to let you control what level of breakage to overcome) and another command for transforming "valid" files (with a set of flags to let you control what additional features to add/remove).

jhurliman, 2022-09-22:
@defunctzombie what would the expected behavior be if you fed an invalid file into a command expecting a valid file, or a valid file into a command expecting an invalid file? It seems like the separate commands would functionally do the same thing, but with a warning or error if your file didn't match the valid/invalid expectation?

defunctzombie, 2022-09-23:
> invalid file into a command expecting a valid file

Error?

> valid file into a command expecting an invalid file

Depends on the flags you want to run. If the command is to repair a file there would be nothing to repair?

Maybe they do degrade down into similar flows for some files tho it does also seem like theres a difference to repairing and expecting certain bad state to patch over vs transforming and expecting certain well-behaved files?

PR #599 (2022-09-28): cli: add recover & compress commands
**Public-Facing Changes**
Added command `mcap recover` for recovering data from truncated MCAP files.
Added command `mcap compress` for changing the compression in an MCAP file.

**Description**
Both new commands share core implementation with `mcap filter`, with different implied options.

Resolves #481 
Resolves #446
commit 0714ada4 (2022-09-28): cli: add recover & compress commands (#599)

**Public-Facing Changes**
Added command `mcap recover` for recovering data from truncated MCAP files.
Added command `mcap compress` for changing the compression in an MCAP file.

**Description**
Both new commands share core implementation with `mcap filter`, with different implied options.

Resolves #481 
Resolves #446
closed 2022-09-28 by jtbandes