foxglove/mcap #490: Review quality bar across each language

amacneil, 2022-07-25:
We should document level of support for each language:
- Document tier of support
- Review supported features (e.g. indexed reads, ordered reads) and standardize these for Tier 1 supported languages

jtbandes, 2022-08-01:
Note that there are multiple levels of ordered reading. I believe some readers support chunks being out of order, but don't make an effort to reorder/merge messages across chunks that overlap.

defunctzombie, 2022-08-01:
Adrian: make a matrix of features

james-rms, 2022-08-04:
Feature matrix here: https://www.notion.so/foxglovehq/Library-Feature-Matrix-43bc4544abf3463884fb719ec449b264

closed 2022-08-04 by james-rms

amacneil, 2022-08-04:
👍 we should move this to public documentation once we have the website up