foxglove/mcap #492: Backwards conversion: MCAP to rosbag1

Hommus, 2022-07-26:
I would like to use the MCAP CLI to convert data from .mcap to .bag (rosbag1) for the purpose of data verification. Thank you!

amacneil, 2022-07-28:
Seems like a nice idea, assuming that the mcap file contains only ROS 1 channels.

mica-angeli, 2023-03-08:
Just wanted to comment that my company could use this feature as well

jtbandes, 2023-03-08:
> for the purpose of data verification

What kind of verification are you hoping to perform on the .bag file?

wkalt, 2023-03-08:
assuming the objective is to be able to go bag -> mcap -> bag and get an identical hashing bag out, some complications to consider will be,
* compression implementations will not always emit the same bytes for the same inputs. I think identical hashes are only likely to come out on uncompressed bag/mcap.
* connection ID numbering would need to be preserved going from bag -> mcap. In theory this could be an issue since bags use a uint32 and mcap uses a uint16. In fact, this is technically a problem in conversion today. In practice most bags have under uint16 (65535) connections.
* the bag header record reserves some empty space for tooling to fill in. Usually this space is just empty. I'm not sure if tooling actually does fill it in. We would need to figure out some place to stuff this in the mcap file though.