foxglove/mcap #496: `mcap convert` should support db3 files with IDL schemas

amacneil, 2022-07-28:
Pre-requisite for [Support MCAP files with IDL schema encoding in Foxglove Studio](https://github.com/foxglove/studio/issues/3968).

We should define a way to represent ROS 2 IDL schemas in MCAP, and support these in `mcap convert`.

wkalt, 2022-07-28:
https://github.com/foxglove/mcap/blob/main/go/ros/ros2db3_to_mcap.go#L247

I think we will just need to parameterize the encoding here, currently hardcoded to "msg".

What should the behavior be if some schemas are present as MSG and others only as IDL? I think the best course of action is probably to add flags for `--schema-encoding` and `--allow-mixed-schema-encodings`. The --schema-encoding flag can default to msg, and fail if any schema with the desired encoding is not found, unless the --allow-mixed-schema-encodings flag is also specified in which case it will just be an ordered preference.

wkalt, 2022-07-28:
another question to address, is how are IDL schemas combined during dependency resolution. In msg files with use === separators. Is there another convention in IDL?

amacneil, 2022-07-28:
I don't think it should be a CLI flag.

We need to do a bit more research, but my understanding of ROS 2 is that each publisher/subscriber can choose to use either `msg` or `idl`. So we should automatically store the correct format depending on what was actually used for that topic.

@emersonknapp probably knows more about this.

wkalt, 2022-07-29:
not sure - within a db3 file at least, I don't see a way to get back msg vs idl. I'd expect to see it in the topics table if it were there, but that looks like this (an example from the rosbag2 repo):

```
id|name|type|serialization_format|offered_qos_profiles
1|AAA|std_msgs/msg/String|cdr|
2|BBB|std_msgs/msg/String|cdr|
3|CCC|std_msgs/msg/String|cdr|
4|DDD|std_msgs/msg/String|cdr|
5|EEE|std_msgs/msg/String|cdr|
6|FFF|std_msgs/msg/String|cdr|
7|GGG|std_msgs/msg/String|cdr|
8|HHH|std_msgs/msg/String|cdr|
```

the metadata.yaml also doesn't give any indication of it either.