foxglove/mcap #501: `mcap info` shows incorrect start/end timestamps for my file

jhurliman, 2022-08-02:
**Description**

When running `mcap info` on [whattimeisit.mcap.zip](https://github.com/foxglove/mcap/files/9238763/whattimeisit.mcap.zip), the following incorrect start/end/duration info is shown:

```
$ mcap info whattimeisit.mcap
library: python-mcap-ros1-support
profile: ros1
messages: 100
duration: -460947h7m35.322474798s
start: 2022-08-02T11:07:45.222474798+08:00 (1659409665.222474798)
end: 1970-01-01T08:00:09.9+08:00 (9.900000000)
compression:
	zstd: [1/1 chunks] (80.31%)
channels:
  	(1) /tf  100 msgs (0.00 Hz)   : tf2_msgs/TFMessage [ros1msg]
attachments: 0
```

Studio shows correct start/end timestamps for this file.

- Version: v0.0.13
- Platform: MacOS

jtbandes, 2022-08-02:
The message start/end time in Statistics appear to be swapped from the chunk index:

[image]

[image]

[image]

jtbandes, 2022-08-02:
I don't think this would be considered an issue with `mcap info`, unless we want to start adding error-detection features to it, but those belong in `mcap doctor`. I would consider it a bug that `mcap doctor` doesn't produce any errors when run on this file.

jtbandes, 2022-08-02:
Studio and the Quick Look preview for whatever reason use the chunk start/end times instead of the one from Statistics:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/IterablePlayer/Mcap/McapIndexedIterableSource.ts#L37-L46

https://github.com/foxglove/studio/blob/main/desktop/quicklook/getIndexedMcapInfo.ts#L66-L77

PR #507 (2022-08-03): doctor: check statistics start/end time
**Public-Facing Changes**
Updated `mcap doctor` to check the correctness of Statistics message_start_time/message_end_time.

**Description**
Fixes #501.

Reports new errors on the file from #501:
```
Examining /Users/jacob/Downloads/whattimeisit.mcap
Statistics has message start time 1659409665222474798, but the minimum chunk start time is 0
Statistics has message end time 9900000000, but the maximum chunk end time is 1659409665222474798
```
commit 7f8ac8f1 (2022-08-03): doctor: check statistics start/end time (#507)

**Public-Facing Changes**
Updated `mcap doctor` to check the correctness of Statistics message_start_time/message_end_time.

**Description**
Fixes #501.

Reports new errors on the file from #501:
```
Examining /Users/jacob/Downloads/whattimeisit.mcap
Statistics has message start time 1659409665222474798, but the minimum chunk start time is 0
Statistics has message end time 9900000000, but the maximum chunk end time is 1659409665222474798
```
closed 2022-08-03 by jtbandes