foxglove/mcap #505: python: statistics/chunk start/end accounting assumes in-order file

wkalt, 2022-08-02:
In the python writer I think we are making a couple assumptions that can lead to incorrect statistics at the end of the file.

I think this code has two bugs - it prevents the statistics record from legitimately starting at zero if subsequent messages are recorded, and also assumes the messages are in chronological order:
https://github.com/foxglove/mcap/blob/main/python/mcap/mcap/mcap0/writer.py#L173,L174

For chunks I think we are doing similar, assuming the first message in a chunk is also the first in time:
https://github.com/foxglove/mcap/blob/main/python/mcap/mcap/mcap0/writer.py#L416,L417

wkalt, 2022-08-02:
note that with the current code, writing a zero-stamped message mid-file will also mess with the file statistics

jtbandes, 2022-08-02:
What were you showing with the 2nd link? I think those times come from here, and this looks more correct: https://github.com/foxglove/mcap/blob/main/python/mcap/mcap/mcap0/chunk_builder.py#L30-L33

wkalt, 2022-08-02:
@jtbandes yeah, reading that again I think the chunk builder is correct. It's just the statistics that have the issue.

closed 2022-08-02 by jtbandes