foxglove/mcap #535: `mcap doctor` reports no errors for corrupt mcap file

defunctzombie, 2022-08-17:
**Description**
> NOTE: I _believe_ this file is corrupt based on not being able to open it in studio with a bad magic error. I looked at the file in a hex viewer and the start of the file is ok - it is somewhere later in reading the file that Studio fails (possibly when it thinks there should be a footer record).

The file is our converted nuscenes.bag -> mcap located here: https://drive.google.com/drive/folders/10TFUJgvhOPqpSQ-q8tSbQum1CX33aO3L

When I run `mcap doctor` on this file there are no errors. But opening in Studio shows this error:

[image]

Studio uses the typescript reader so I imagine the result would be similar outside of studio for direct users of the typescript reader.

jtbandes, 2022-08-17:
Note that the "bad magic" that studio complains about is located _near_ the end of the file. And it seems like the file does not end with the expected trailing magic.

[image]

defunctzombie, 2022-08-17:
@jtbandes that byte string also occurs in many other places in the file (at least that is what I saw in my hex viewer).

jtbandes, 2022-08-17:
Oh yeah, you're right it does. I was just looking near the end because I wanted to see the trailing magic.

PR #545 (2022-08-18): doctor: report missing DataEnd/Footer
**Public-Facing Changes**
`mcap doctor` now reports files missing their DataEnd or Footer records.

**Description**
Fixes #535 
Also updates the TypeScript `validate` script to report extra bytes remaining after parsing.

Ideally doctor would also validate trailing magic. I believe today it does not.
commit 366763e6 (2022-08-18): doctor: report missing DataEnd/Footer (#545)

**Public-Facing Changes**
`mcap doctor` now reports files missing their DataEnd or Footer records.

**Description**
Fixes #535 
Also updates the TypeScript `validate` script to report extra bytes remaining after parsing.

Ideally doctor would also validate trailing magic. I believe today it does not.
closed 2022-08-18 by jtbandes