foxglove/mcap #536: `mcap convert` fails to convert this bag file

defunctzombie, 2022-08-17:
Run `mcap convert` on this bag file: https://librealsense.intel.com/rs-tests/TestData/stairs.bag

The resulting `mcap info` has a very incorrect duration compared to the start/end shown by `rosbag info`.

```
(ros_env) ➜  Downloads rosbag info stairs.bag
path:         stairs.bag
version:      2.0
duration:     10.2s
start:        Dec 31 1969 16:00:00.00 (0.00)
end:          Dec 31 1969 16:00:10.23 (10.23)
size:         390.3 MB
messages:     926
compression:  lz4 [222/222 chunks; 55.60%]
uncompressed: 701.7 MB @ 68.6 MB/s
compressed:   390.2 MB @ 38.1 MB/s (55.60%)
types:        diagnostic_msgs/KeyValue  [cf57fdc6617a881a88c16e768132149c]
              geometry_msgs/Transform   [ac9eff44abf714214112b05d54a3cf9b]
              realsense_msgs/StreamInfo [311d7e24eac31bb87271d041bf70ff7d]
              sensor_msgs/CameraInfo    [c9a58c1b0b154e0e6da7578cb991d214]
              sensor_msgs/Image         [060021388200f6f0f447d0fcd9c64743]
              std_msgs/UInt32           [304a39449588c7f8ce2df6e8001c5fce]
topics:       /device_0/info                                  7 msgs    : diagnostic_msgs/KeyValue
              /device_0/sensor_0/Depth_0/image/data         146 msgs    : sensor_msgs/Image
              /device_0/sensor_0/Depth_0/image/metadata     438 msgs    : diagnostic_msgs/KeyValue
              /device_0/sensor_0/Depth_0/info                 1 msg     : realsense_msgs/StreamInfo
              /device_0/sensor_0/Depth_0/info/camera_info     1 msg     : sensor_msgs/CameraInfo
              /device_0/sensor_0/Depth_0/tf/0                 1 msg     : geometry_msgs/Transform
              /device_0/sensor_0/info                         1 msg     : diagnostic_msgs/KeyValue
              /device_0/sensor_0/property                    12 msgs    : diagnostic_msgs/KeyValue
              /device_0/sensor_1/Color_0/image/data          75 msgs    : sensor_msgs/Image
              /device_0/sensor_1/Color_0/image/metadata     225 msgs    : diagnostic_msgs/KeyValue
              /device_0/sensor_1/Color_0/info                 1 msg     : realsense_msgs/StreamInfo
              /device_0/sensor_1/Color_0/info/camera_info     1 msg     : sensor_msgs/CameraInfo
              /device_0/sensor_1/Color_0/tf/0                 1 msg     : geometry_msgs/Transform
              /device_0/sensor_1/info                         1 msg     : diagnostic_msgs/KeyValue
              /device_0/sensor_1/property                    14 msgs    : diagnostic_msgs/KeyValue
              /file_version                                   1 msg     : std_msgs/UInt32
```

```
(ros_env) ➜  Downloads mcap info stairs.mcap
library: mcap go #(devel)
profile: ros1
messages: 926
duration: 46h31m14.671963831s
start: 0.000
end: 167474.672
compression:
	zstd: [66/66 chunks] (62.52%)
channels:
  	(0) /file_version                                   1 msgs (0.00 Hz)   : std_msgs/UInt32 [ros1msg]
  	(1) /device_0/sensor_0/Depth_0/info                 1 msgs (0.00 Hz)   : realsense_msgs/StreamInfo [ros1msg]
  	(2) /device_0/sensor_0/Depth_0/info/camera_info     1 msgs (0.00 Hz)   : sensor_msgs/CameraInfo [ros1msg]
  	(3) /device_0/info                                  7 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(4) /device_0/sensor_0/info                         1 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(5) /device_0/sensor_0/property                    12 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(6) /device_0/sensor_1/info                         1 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(7) /device_0/sensor_1/property                    14 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(8) /device_0/sensor_0/Depth_0/image/data         146 msgs (0.00 Hz)   : sensor_msgs/Image [ros1msg]
  	(9) /device_0/sensor_0/Depth_0/image/metadata     438 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(10) /device_0/sensor_0/Depth_0/tf/0                1 msgs (0.00 Hz)   : geometry_msgs/Transform [ros1msg]
  	(11) /device_0/sensor_1/Color_0/info                1 msgs (0.00 Hz)   : realsense_msgs/StreamInfo [ros1msg]
  	(12) /device_0/sensor_1/Color_0/info/camera_info    1 msgs (0.00 Hz)   : sensor_msgs/CameraInfo [ros1msg]
  	(13) /device_0/sensor_1/Color_0/image/data         75 msgs (0.00 Hz)   : sensor_msgs/Image [ros1msg]
  	(14) /device_0/sensor_1/Color_0/image/metadata    225 msgs (0.00 Hz)   : diagnostic_msgs/KeyValue [ros1msg]
  	(15) /device_0/sensor_1/Color_0/tf/0                1 msgs (0.00 Hz)   : geometry_msgs/Transform [ros1msg]
attachments: 0
```

wkalt, 2022-08-29:
https://github.com/foxglove/mcap/issues/501 was the issue this reminded me of, but it may not be related at all

james-rms, 2022-09-14:
I took a look at this, it appears `rosbag` is wrong, not `mcap`. There are two messages very early on with:

| topic | secs | nsecs
| --- | --- | --- |
| /device_0/sensor_0/Depth_0/info |  167474 | 671963832 |
| /device_0/sensor_0/Depth_0/info/camera_info | 167474 | 671963832 |

the rest of the timestamps start from 0. The input data is odd, but there's nothing strictly wrong with some messages having odd timestamps.

closed 2022-09-14 by james-rms

defunctzombie, 2022-09-14:
👏 Thanks for investigating!