foxglove/mcap #585: Typescript reader example

defunctzombie, 2022-09-17:
I was working on an experiment that required loading reading an mcap file in a web browser. Looking at the examples/guides/docs I didn't find anything to figure out how to read an mcap file in the browser. Ended up having to read Studio code.

It would be great to provide basic read/write examples for all the supported languages and the common environments they are used in.

jtbandes, 2022-11-11:
We have https://github.com/foxglove/mcap/blob/main/typescript/examples/validate/scripts/validate.ts. Are you looking for something more focused on simple reading? Are you looking for a browser-based example that doesn't use node fs?

defunctzombie, 2022-11-11:
@jtbandes validate is full of other noise - I was thinking something close to: https://github.com/foxglove/mcap/tree/main/typescript/examples/basicwriter

gkjohnson, 2024-05-23:
I'm interested in a simple mcap file read example for Javascript / Typescript, as well. Specifically I'm interested in time-sequentially iterating over all messages in the loaded mcap file. The [API reference](https://mcap.dev/docs/typescript/) looks like it's primarily a list of classes and member definitions but no behavior or usage descriptions so I'm not exactly sure where to start.

I'm only planning to work with this in node, at the moment, but I'm imagining the core code shouldn't different too significantly? Just swapping [BlobReadable](https://mcap.dev/docs/typescript/classes/_mcap_browser.BlobReadable) and [FileReadable](https://mcap.dev/docs/typescript/classes/_mcap_nodejs.FileHandleReadable)?

As always - thanks for making your work open source, guys!

defunctzombie, 2024-05-23:
The simplest example I can point to is this validate script which uses an Indexed mcap reader: https://github.com/foxglove/mcap/blob/main/typescript/examples/validate/scripts/validate.ts#L226-L237

The messages are read here: https://github.com/foxglove/mcap/blob/main/typescript/examples/validate/scripts/validate.ts#L236