foxglove/mcap #606: Typescript: lodash should be a dependency, not devDependancy

jbcpollak, 2022-09-23:
**Description**

It looks like lodash is a devDependency but is used in production.

- Version: "@mcap/core": "^0.2.2",
- Platform: Typescript

**Steps To Reproduce**

importing `@mcap/core` results in an import error:

```
import { Mcap0Writer, IWritable } from "@mcap/core";
```

**Expected Behavior**

No errors on import

**Actual Behavior**

```
Error: Cannot find module 'lodash'
Require stack:
- /project/node_modules/@mcap/core/dist/cjs/src/pre0/parse.js
- /project/node_modules/@mcap/core/dist/cjs/src/pre0/McapPre0Reader.js
- /project/node_modules/@mcap/core/dist/cjs/src/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/project/node_modules/@mcap/core/src/pre0/parse.ts:1:1)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1157:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/project/node_modules/@mcap/core/dist/cjs/src/pre0/parse.js',
    '/project/node_modules/@mcap/core/dist/cjs/src/pre0/McapPre0Reader.js',
    '/project/node_modules/@mcap/core/dist/cjs/src/index.js'
  ]
}
```

jhurliman, 2022-09-23:
Fixed in https://github.com/foxglove/mcap/pull/597

closed 2022-09-23 by jhurliman

jbcpollak, 2022-09-23:
🎉 thanks, I _did_ search for issues and PRs but guess I forgot to check for _closed_ issues.