foxglove/mcap #644: Standardize default chunk size across official libraries

amacneil, 2022-10-10:
From @wkalt:
> yeah, there is some variance. the libs are 1MB for python and go, 768kb for C++, and 10MB for swift
> with the go CLI using 8MB

We should decide on a default (based on various trade-offs, or best guess for now) and standardize this across all official library implementations (those in this repo).

jhurliman, 2022-10-10:
There are two separate categories to consider, library defaults and tooling defaults. The quote above mixes these two categories. For libraries, I agree it makes sense to standardize the chunk size (I propose 1MB). But for tooling, there are different tools for different purposes. The 768KB default for the rosbag2 plugin comes from the ROS1 default chunk size, which was found by the community to be a tradeoff between compression efficiency and data loss in real-time recording scenarios. The 8MB default in the mcap cli is a tradeoff between compression efficiency and memory usage during decoding.

While addressing this ticket, let's make sure we are modifying library defaults and not tooling defaults.

amacneil, 2022-10-10:
Understood re libraries vs tooling.

> The 768KB default for the rosbag2 plugin comes from the ROS1 default chunk size, which was found by the community to be a tradeoff between compression efficiency and data loss in real-time recording scenarios

I'm curious if there is any known documentation around this, or if someone just picked something in the early days of ROS1 and it stuck.