foxglove/mcap #683: Remove `profile` feature?

amacneil, 2022-10-26:
The spec currently contains a `profile` field in the `Header` message, with documented suggested values in [the appendix](https://mcap.dev/specification/appendix.html#well-known-profiles).

This feature is only used for one feature in Foxglove Studio (tf normalization), but Data Platform currently [deletes the profile](https://github.com/foxglove/studio/issues/4690). However, there are probably better ways to handle that particular problem.

Otherwise, this field doesn't make a lot of sense to exist at the file level - it's perfectly valid to create (or merge) a file containing channels with different encodings. So I suggest removing it and making the `profile` field reserved going forward.

amacneil, 2022-10-26:
cc @defunctzombie @jhurliman @wkalt @james-rms @jtbandes

jhurliman, 2022-10-26:
Beyond frame_id normalization and parameter normalization, the ros profiles tell us that certain parameters are semantically meaningful. This is used to detect robot restarts when connected to a live robot, and to make URDFs work. There will likely be more profile usage as we get closer to feature parity with rviz, or support other robot frameworks in the future.

The profile isn't about instructing us how to treat individual data channels; that's why it's at the file level and not the channel level. It conveys global semantics across a session of a robot. You're correct that MCAP does not support recording from two separate robots that are running different robotics frameworks but recording into the same file while *also* capturing the global semantics of both robots simultaneously.

defunctzombie, 2022-10-27:
The `profile` concept is influenced by its peer concept in the video coding space: https://en.wikipedia.org/wiki/Advanced_Video_Coding#Profiles.

MCAP files are generic containers which could contain schemas and messages across many different formats. When I want to adopt MCAP for my framework I want to limit the universe of things I need to support so I can create tooling specific to my framework. The profile allows me to specify that my tooling is "feature complete" by limiting the universe of what a valid mcap file for my framework contains.

A specific example would be the `ros2 bag play -s mcap` command. If I load up an MCAP file with JSON messages or Protobuf messages and try to play the file it won't work. Why not? These are valid MCAP files? Should I open a feature request to play protobuf data using `ros2 bag play`? That's one approach. The alternative is to say that we only support playing MCAP files with `ros2msg`, `ros2idl` and `cdr` schemas and data. `profile` is provided as a means to capture this constraint in a single concept.

I can state that my tools support MCAP files with the `ros2` profile and understand this aligns with what my framework has defined as a "valid" MCAP file it will produce and my tooling can support.

There may be other ways to capture the above idea but the overall concept of adapting a generic and flexible format to your more narrow application domain remains.

amacneil, 2022-10-27:
The video example is interesting, but how they are using `profile` is more similar to our `message.encoding` (they seem to specify actual encoding details, not just semantics).

Re `ros2 bag play`, I think it is a good discussion case. Do we intend for it to refuse to play a file unless the `profile` is set to `ros2`? Or is it just going to play the `cdr` channels it knows about, and ignore/warn about any other channels? (I asked this question internally today and was told that the answer is neither - we currently just try to play all the channels regardless of their encoding).

defunctzombie, 2022-10-27:
> Do we intend for it to refuse to play a file unless the `profile` is set to `ros2`? Or is it just going to play the `cdr` channels it knows about, and ignore/warn about any other channels? (I asked [...]

Either one feels valid to me. If I was an author of a player for a framework like ros and someone came along and asked to have a feature to play their JSON or Protobuf data I'd probably prefer to have something I can point at that says "That is not a valid ROS MCAP file" even tho it is a valid MCAP file.

jhurliman, 2023-02-16:
Now that the `profile` field is actively used in Foxglove Studio, in the websocket protocol, and for in-house usage to identify the source framework, this would be a breaking change to MCAP 1.0. Closing for now, we can resurface if/when MCAP 2.0 comes around.

closed 2023-02-16 by jtbandes