foxglove/mcap #709: RFC: Change the "ros2" profile to allow CDR-encoded messages with no preceding Schema record

jhurliman, 2022-11-04:
##Overview##

Today, rosbag2+mcap_storage will make a best-effort attempt to locate a `.msg` or `.idl` file given a datatype and create a Schema record from it. If the message definition does not exist on the local disk, or AMENT_PREFIX_PATH is missing or incorrect, or the ament index is incomplete/missing/corrupt, or there is a read access permission failure, the Schema record will not be created. To reflect that sometimes CDR channels will be recorded under the "ros2" profile without an associated Schema, we should change the "ros2" profile spec to reflect this.

##Impact##

Downstream readers should be updated to account for this new definition of the "ros2" profile. If a "cdr" Channel is encountered with no accompanying Schema, the reader may choose to search a local cache such as built-in message definitions or an ament index on disk to try and locate a `.msg` or `.idl` matching the Channel schema name (datatype). If no message definition is found or the reader chooses not to implement this fallback, decoding cannot occur for that channel. It's up to the application to choose how to handle this. Examples may include proceeding with decoding of all other channels, exiting with an error, performing some form of playback but treating the message payload as opaque, etc.

jtbandes, 2022-11-04:
> If a "cdr" Channel is encountered with no accompanying Schema, the reader may ... try and locate a .msg or .idl matching the Channel schema name

How can there be a schema name if there's no schema?

jhurliman, 2022-11-04:
> > If a "cdr" Channel is encountered with no accompanying Schema, the reader may ... try and locate a .msg or .idl matching the Channel schema name
> 
> How can there be a schema name if there's no schema?

I'm using the terms "schema name" and "datatype" interchangeably here. A ROS 2 system knows the association of "topic name" -> "datatype" before recording ever starts, and it's the datatype that is used in the ament index lookup. That same datatype is written into the Schema record in the `schema_name` field if a Schema record is successfully created.