foxglove/mcap #718: Making `BuildFileDescriptorSet` part your mcap API

armuren, 2022-11-11:
I may have misunderstood something here.  In your cpp protobuf example, you have a function [BuildFileDescriptorSet](https://github.com/foxglove/mcap/blob/main/cpp/examples/protobuf/BuildFileDescriptorSet.cpp#L10) that does not seem to be dependent on of the messages themselves.  

So then each different project would just copy/paste these the header and the cpp files.

Why isn't this function part of your mcap c++ API?  As a consumer I would hope to not have to worry about this kind of function and potentially having to change it when you put out a new version of mcap.

Or have I missed something here??

jtbandes, 2022-11-11:
Thanks for your interest. This is a good question. The main reason this is not exported as part of the mcap library is that it depends on protobuf, and MCAP is meant to be serialization-agnostic. We do not want all MCAP users to be required to take on a protobuf dependency for their project to build. That said, I agree that it would be nice to publish the function somehow so that people do not have to copy/paste it. Out of curiosity, how are you getting mcap into your project — are you using conan?

As for having to change the function in the future, that would not be expected, because this function generates a well-known FileDescriptorSet object that should match the output of `protoc --descriptor_set_out`. That is not something that will change as MCAP evolves.

armuren, 2022-11-11:
Thanks @jtbandes ! I didn't think about the dependency issue and that's ofc a great point.

We are using bazel to build our project and so I am still trying to figure out the best way to integration MCAP as a third_party dependency.  It maybe a end up being a loaded local library, or may live next the code that uses it.  Not sure yet, any advice on this?

jtbandes, 2022-11-11:
How do you integrate your other third party dependencies normally?

armuren, 2022-11-11:
Compiled binaries to `/usr/local/lib` and loaded as bazel `new_local_repository`