foxglove/mcap #719: Some boilerplate is required to set up decompression handlers (see https://github.com/foxglove/ws-protocol/pull/270)

jtbandes, 2022-11-11:
Some boilerplate is required to set up decompression handlers when reading an MCAP file. In https://github.com/foxglove/ws-protocol/pull/270 I copy/pasted some of the code we use for this from another repo. We have similar code both here and in foxglove/studio. These should be published to NPM so they can be reused.

Library name ideas:
- `@mcap/compression-support`
- `@mcap/wasm-compression-support`

jhurliman, 2023-02-17:
There is also boilerplate needed to define readers for node.js and web file handles. Could those be rolled into the same package?

jtbandes, 2023-02-17:
Depends how granular/hygienic you want the dependencies to be. Would a web user import the same package as a nodejs user?

jhurliman, 2023-02-17:
I was thinking something like https://github.com/foxglove/ros2/blob/main/nodejs.js

PR #868 (2023-11-21): Add @mcap/support, @mcap/nodejs, and @mcap/browser libraries
**Public-Facing Changes**
- [TypeScript] Added a new package `@mcap/support`. This package provides decompression functions for zstd, lz4, and bz2, and utility functions for working with protobuf descriptors.
- [TypeScript] Added a new package `@mcap/nodejs`. This package provides IReadable and IWritable implementations for Node.js `FileHandle`, reducing boilerplate for reading or writing MCAP files.
- [TypeScript] Added a new package `@mcap/browser`. This package provides an IReadable implementation for the browser `Blob` (`File`) t [... cut]
commit 16bc9a8c (2023-11-21): Add @mcap/support, @mcap/nodejs, and @mcap/browser libraries (#868)

**Public-Facing Changes**
- [TypeScript] Added a new package `@mcap/support`. This package
provides decompression functions for zstd, lz4, and bz2, and utility
functions for working with protobuf descriptors.
- [TypeScript] Added a new package `@mcap/nodejs`. This package provides
IReadable and IWritable implementations for Node.js `FileHandle`,
reducing boilerplate for reading or writing MCAP files.
- [TypeScript] Added a new package `@mcap/browser`. This package
provides an IReadable implementation for the browser `Blob` (` [... cut]
closed 2023-11-21 by jtbandes