foxglove/mcap #721: c++: assert (written == len) is not informative error message for write failure

james-rms, 2022-11-13:
**Description**
Currently we assert that the write succeeds in `FileWriter`, here:
https://github.com/foxglove/mcap/blob/d850fc5938e67305a93c58d907a1a20f394cb8ed/cpp/mcap/include/mcap/writer.inl#L55

This is an error that can occur in production if the underlying storage medium fills up, and the user needs some kind of trail to debug why. 
- Version: v0.6.0
- Platform: Ubuntu Linux

**Steps To Reproduce**
Create a tmpfs with a limited max size, try writing past its size limit.

**Expected Behavior**
Some kind of "storage is full" error

**Actual Behavior**
"Assertion failed (size == written)"

jhurliman, 2022-11-13:
Is the storage medium filling up the only possible cause of this error?

james-rms, 2022-11-13:
No, this generalizes to all the errors that could result from calling `fwrite`. On linux, there's a list here: https://man7.org/linux/man-pages/man3/fputc.3p.html

jhurliman, 2022-11-14:
Got it. Is the idea that we check for `size != written` and then call `errno()` + `strerror()`?

james-rms, 2022-11-14:
Yes, modulo thread-safety concerns. We may need some OS-specific stuff here.

jhurliman, 2022-11-16:
`errno()` on Linux uses thread-local storage. I don't know about other platforms, and it will need some sort of `#ifdef` for Windows I imagine.