foxglove/mcap #727: rust: no way to limit chunk size automatically

james-rms, 2022-11-16:
**Description**

Right now if you write an MCAP with the rust client, you will end up with one large chunk full of all of your messages unless you explicitly call `finish_chunk`. This can result in giant chunks that are larger than `2**31` bytes. This can cause errors in readers which assume that a chunk can be held in memory all at once.

- Version: `releases/rust/v0.5.0`
- Platform: All

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

neilisaac, 2022-11-17:
@james-rms finish_chunk is currently private, and chunks are buffered in memory, so this is a critical issue.

james-rms, 2022-11-17:
@neilisaac for now you can call `flush()` which is public and should do what you need.

PR #754 (2022-11-30): rust: writer automatically breaks chunks at size
**Public-Facing Changes**

Adds a new writer option, `chunk_size`, which will automatically break up chunks at a target (uncompressed) size.

**Description**

Fixes #727
closed 2022-11-30 by james-rms