foxglove/mcap #732: specification: metadata records must appear outside chunks

wkalt, 2022-11-17:
We have a disclaimer to this effect about attachments. The same needs to apply for metadata records, if the record is to be referenced by metadata index records. A metadata record that appears within a chunk cannot be referenced by a metadata index record today, which relies on a global file offset - compression will mess with a reader's ability to actually make use of an offset into compressed data.

There is potentially some leeway if there is a reason to permit recording unindexed metadata records into chunks. To keep things simple, seems best to me that the spec ban this.

jtbandes, 2022-11-17:
> A Chunk contains a batch of Schema, Channel, and Message records.

https://github.com/foxglove/mcap/tree/main/docs/specification#chunk-op0x06

james-rms, 2022-11-28:
@wkalt do you still think the metadata record needs this disclaimer?

closed 2023-01-09 by james-rms