foxglove/mcap #745: Small bug in documentation

ggaessler, 2022-11-22:
In https://mcap.dev/guides/python/ros2.html the Python code includes the line 
```
        print(f"{topic} ({type(msg).__name__}) [{timestamp}]: '{msg.data}'")
```
asssuming that only ROS msgs would be contained in the bag. However one of the points of MCAP is to not only record ROS msgs, but also for example ROS logs. And logs (so `rcl_interfaces.msg._log.Log`) do not contain an `data` attribute, at least in Galactic. So that `AttributeError` exception should be caught. 

- Version: documentation as of today

**Steps To Reproduce**
Copy code from documentation and run it on a bag containing logs.
```
import argparse

import rosbag2_py
from rclpy.serialization import deserialize_message
from rosidl_runtime_py.utilities import get_message

def read_messages(input_bag: str):
    reader = rosbag2_py.SequentialReader()
    reader.open(
        rosbag2_py.StorageOptions(uri=input_bag, storage_id="mcap"),
        rosbag2_py.ConverterOptions(
            input_serialization_format="cdr", output_serialization_format="cdr"
        ),
    )

    topic_types = reader.get_all_topics_and_types()

    def typename(topic_name):
        for topic_type in topic_types:
            if topic_type.name == topic_name:
                return topic_type.type
        raise ValueError(f"topic {topic_name} not in bag")

    while reader.has_next():
        topic, data, timestamp = reader.read_next()
        msg_type = get_message(typename(topic))
        msg = deserialize_message(data, msg_type)
        yield topic, msg, timestamp
    del reader

def main():
    parser = argparse.ArgumentParser(description=__doc__)
    parser.add_argument(
        "input", help="input bag path (folder or filepath) to read from"
    )

    args = parser.parse_args()
    for topic, msg, timestamp in read_messages(args.input):
        print(f"{topic} ({type(msg).__name__}) [{timestamp}]: '{msg.data}'")

if __name__ == "__main__":
    main()
```
source: https://mcap.dev/guides/python/ros2.html

**Expected Behavior**
No error.

**Actual Behavior**
```
Traceback (most recent call last):
  File "./test_mcap.py", line 48, in <module>
    main()
  File "./test_mcap.py", line 45, in main
    print(f"{topic} ({type(msg).__name__}) [{timestamp}]: '{msg.data}'")
AttributeError: 'Log' object has no attribute 'data'
```

jtbandes, 2022-12-01:
Thanks for the report. Would you be interested in creating a PR to update the documentation? The source is here: https://github.com/foxglove/mcap/blob/main/docs/guides/python/ros2.md

ggaessler, 2022-12-08:
Sorry for the late reply. Yes, I would be. Unfortunately due to company guidelines I'm just allowed to contribute to a fixed list of repositories/projects and this also applies to trivial PRs such as this one. However mcap will be added to this list with the next revision, probably end of January. So either someone wants to fix it right now or I'll do it once I'm allowed to.

defunctzombie, 2023-01-10:
What would be a more appropriate thing to print? Should we print `{msg}` instead?

PR #824 (2023-02-21): docs: ros2: handle case using rosbag2_py for non-string msgs
**Public-Facing Changes**
ROS2 example now handles non-String messages. Leads to a smoother intro for ROS 2 users with existing bags.

**Description**

Fixes #745
closed 2023-02-21 by james-rms