foxglove/mcap #755: Distribute mcap CLI as .deb package

james-rms, 2022-11-30:
Copying over from https://github.com/ros-tooling/rosbag2_storage_mcap/issues/47

Proposed plan from @amacneil:
> Installing it as a ros2 mcap command might be a good option to avoid conflict with an Ubuntu universe debian package.
> I personally think the developer experience of having the CLI available by default is more important than the reduced footprint of one less binary on the robot, given how much else ROS installs.

> But if people feel strongly about reduced runtime robot footprint, then here is a concrete proposal:

- New mcap_cli package that installs the ros2 mcap subcommand.
- Make the mcap_cli package a dependency of rosbag2 metapackage (but not a dependency of rosbag2_storage_mcap), so that it is installed by default (see [existing rosbag2 dependencies](https://index.ros.org/p/rosbag2/#humble-deps))
- People who do not want the mcap CLI can install the specific dependencies of rosbag2 metapackage that they need (as they presumably would already be doing if they care about minimizing on-robot footprint)

jtbandes, 2024-11-27:
Related to #675

ndmmdn, 2025-02-19:
What is the status on this? Is there a way to install mcap via some package manager?

amacneil, 2025-02-19:
You can install using homebrew: `brew install mcap`.

We don't have a debian package - never looked in to what is required to get added upstream. Would be happy for someone to pick this up.