foxglove/mcap #792: Ambiguous wording in the MCAP File Format Specification

sanjaynarayanan-brt, 2023-01-18:
The following wording in the specification of a Chunk is a bit ambiguous -

> All messages in the chunk must reference channels recorded earlier in the file (in a previous chunk or earlier in the current chunk).

It may be interpreted as "all channels must be in a chunk (previous or current)". However, as far as I'm aware, this is not necessary. Channels can be logged standalone outside a chunk, in the Data section. Please correct me if I'm wrong.

james-rms, 2023-01-18:
Thanks for the report @sanjaynarayanan-brt  - the specification currently allows channel, schema, or message records both in chunks and directly in the data section within the same MCAP. however, most MCAP reader implementations won't handle this correctly. I will amend the spec to disallow this case.

sanjaynarayanan-brt, 2023-01-19:
@james-rms Thanks! 

I'm not sure what MCAP reader runs under the hood in Foxglove Studio (specifically for plots), but we write channels and schemas directly into the data section and put messages into chunks, and that seems to work great with plots.

PR #827 (2023-02-22): specification: clarify where channels can be recorded
**Public-Facing Changes**

**Description**

Fixes #792
closed 2023-02-22 by james-rms