foxglove/mcap #802: mcap 0.8.0 tarball SHA1 hash changed

clalancette, 2023-01-30:
**Description**

The ROS 2 vendoring of mcap is failing to build right now.  An example is the CI run here: https://ci.ros2.org/job/ci_linux/18001/console.  In particular, the interesting part is this:

```
00:04:15.931 -- verifying file...
00:04:15.931        file='/home/jenkins-agent/workspace/ci_linux/ws/build/mcap_vendor/_deps/mcap-subbuild/mcap-populate-prefix/src/v0.8.0.tar.gz'
00:04:15.931 -- SHA1 hash of
00:04:15.931     /home/jenkins-agent/workspace/ci_linux/ws/build/mcap_vendor/_deps/mcap-subbuild/mcap-populate-prefix/src/v0.8.0.tar.gz
00:04:15.931   does not match expected value
00:04:15.931     expected: 'b44637791da2c9c1cec61a3ba6994f1ef63a228c'
00:04:15.931       actual: '5f229666c13280af14eeb0485da5100bad606184'
00:04:15.931 -- Hash mismatch, removing...
```

It seems like somehow the tarball SHA1 hash has changed.  We can update the hash in https://github.com/ros2/rosbag2/blob/rolling/mcap_vendor/CMakeLists.txt , but we were wondering if anything has changed in MCAP to have caused this?

Relatedly, we think it would be better to upload a tarball that the project has built, rather than relying on GitHub to auto-generate the tarballs for releases.  The reason is that the auto-generated tarballs aren't guaranteed to always be the same, so we can side-step the issue by just having an uploaded tarball in the "Releases" section.

- Version: 0.8.0
- Platform: All, but tested on Ubuntu 22.04

**Steps To Reproduce**

Try to build mcap_vendor from ROS 2, like in https://ci.ros2.org/job/ci_linux/18001

**Expected Behavior**

Successful download and build.

clalancette, 2023-01-30:
And it turns out that GitHub recently changed their compression: https://github.blog/changelog/2023-01-30-git-archive-checksums-may-change/ .  So that argues even more strongly for uploading an artifact during the release process.

james-rms, 2023-02-05:
@clalancette The linked change was reverted, will take this as a a feature request to publish a well-defined compressed source artifact to protect against future breakages.

closed 2023-02-05 by james-rms