foxglove/mcap #810: [C++] Unclear documentation for adding schemas

jarkenau, 2023-02-10:
The documentation for C++ includes a [minimal example](https://github.com/foxglove/mcap/tree/main/cpp) on how to add schemas. For writing a simple `std_msgs/String` it can be done like this:

```C++
...
// Register a Schema
mcap::Schema stdMsgsString("std_msgs/String", "ros1msg", "string data");
writer.addSchema(stdMsgsString);

// Register a Channel
mcap::Channel chatterPublisher("/chatter", "ros1", stdMsgsString.id);
writer.addChannel(chatterPublisher);
...
```
Where `string data` describes the layout of the message. How should the string look like for more complex & nested messages, e.g `sensor_msgs/Image`?

jtbandes, 2023-02-10:
The format for ROS schemas is documented here: https://mcap.dev/specification/appendix.html

jarkenau, 2023-02-13:
Thanks for the hint :+1: . But it still won't work as expected. As described in the docs [here](https://mcap.dev/specification/appendix.html#ros1msg-data-format) I used `gendeps --cat` to generate the description and saving `sensor_msgs/Image` now works just fine. If I open the written `.mcap` file in foxglove studio, the Image panel throws the following error `Cannot read properties of undefined (reading 'stamp')`, am I missing something?