foxglove/mcap #822: negative end time reported by `mcap info`, disagrees with duration

jhurliman, 2023-02-18:
**Description**

Repro file: [zfp.mcap.zip](https://github.com/foxglove/mcap/files/10773480/zfp.mcap.zip)

```
$ mcap info zfp.mcap
library: libmcap 1.0.0
profile: 
messages: 300
duration: 2562047h47m16.854775807s
start: 0.000
end: -0.090
compression:
        zstd: [1/1 chunks] [221.93 KiB/3.77 KiB (98.30%)] [0.00 B/sec] 
channels:
        (1) image              100 msgs (0.00 Hz)   : foxglove.CompressedImage [protobuf]    
        (2) camera_info        100 msgs (0.00 Hz)   : foxglove.CameraCalibration [protobuf]  
        (3) image_annotations  100 msgs (0.00 Hz)   : foxglove.ImageAnnotations [protobuf]   
attachments: 0
metadata: 0
```

jtbandes, 2023-02-21:
According to the TS reader, the end time is in 2554:

[image]

jtbandes, 2023-02-21:
[image]

jhurliman, 2023-02-21:
0xfffffffffaa3b800 comes after 0x0!

james-rms, 2023-02-21:
Yeah - figured the same thing, but i also figure we should be able to handle this.

PR #844 (2023-03-07): CLI: info: handle durations > 2**63
### Public-Facing Changes
`mcap info` on files with a duration > 2**63s will have their duration, start and end times printed correctly.

### Description
* fix a bug in decimalTime which would print timestamps > 2^63ns as negative numbers
* fix a bug where very long durations in MCAP would be printed as negative.

For the linked file in #822, returns:
```
$ mcap info
library: libmcap 1.0.0
profile: 
messages: 300
duration: 18446744073.620s
start: 0.000000000
end: 18446744073.619617024
compression:
        zstd: [1/1 chunks] [221.93 KiB/3.77 KiB (98.30%)] [0. [... cut]
closed 2023-03-07 by james-rms