foxglove/mcap #834: `make` command fails in `cpp/`

jhurliman, 2023-02-25:
**Description**

In a clean checkout on a Linux machine, running `make` inside the `cpp/` folder fails:

```
mcap/cpp$ make
...
Setting up libexpat1:amd64 (2.2.9-1ubuntu0.6) ...
Setting up python3.8-minimal (3.8.10-0ubuntu1~20.04.6) ...
Traceback (most recent call last):
  File "/usr/lib/python3.8/py_compile.py", line 215, in <module>
    sys.exit(main())
  File "/usr/lib/python3.8/py_compile.py", line 207, in main
    compile(filename, doraise=True)
  File "/usr/lib/python3.8/py_compile.py", line 172, in compile
    importlib._bootstrap_external._write_atomic(cfile, bytecode, mode)
  File "<frozen importlib._bootstrap_external>", line 186, in _write_atomic
PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.8/__pycache__/__future__.cpython-38.pyc.140704204492448'
dpkg: error processing package python3.8-minimal (--configure):
 installed python3.8-minimal package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3.8-minimal
E: Sub-process /usr/bin/dpkg returned an error code (1)
The command '/bin/sh -c apt-get update &&   apt-get install -y --no-install-recommends --no-install-suggests   ca-certificates   curl   cmake   gnupg   make   perl   python3   python3-pip' returned a non-zero code: 100
```

- Version: `main`
- Platform: Ubuntu 22.04, Docker 20.10.17

james-rms, 2023-03-05:
I'm struggling to replicate this - I started from a fresh VM running ubuntu 22.04, a new clone and an empty docker cache and  successfully built the binaries with `cd cpp && make`.

wkalt, 2023-04-04:
it works for me too

closed 2023-04-04 by wkalt