foxglove/mcap #845: statistics.message_count is not correct in mcap-py

HyokaChen, 2023-03-07:
**Description**
use the mcap-py to extract mcap's info, the statistics.message_count is not correct, below is in python:
[image]

below is mcap cli's result:
[image]

- Version: mcap-py 1.0.1
- Platform: MacOS 13.2.1 Ventura

**Steps To Reproduce**

any mcap package with many topics

**Expected Behavior**
statistics.message_count = 30210

jtbandes, 2023-03-07:
Thank you for the bug report. Would you be interested in creating a pull request to fix the issue?

james-rms, 2023-03-08:
Hi @HyokaChen , are you able to provide the code you used to produce that python output? I don't see any code in the MCAP library itself that would produce a `compression.compression_ratio` value.

jtbandes, 2023-03-08:
It looks like the bug stems from this line of code:

https://github.com/foxglove/mcap/blob/39545dbea9b40fbc55a58df8f4ceecd06150ce49/python/mcap/mcap/records.py#L482

HyokaChen, 2023-03-08:
> It looks like the bug stems from this line of code:
> 
> https://github.com/foxglove/mcap/blob/39545dbea9b40fbc55a58df8f4ceecd06150ce49/python/mcap/mcap/records.py#L482

👍 the loop to extract the Chanel’s message_count override the initial value.

HyokaChen, 2023-03-08:
> Hi @HyokaChen , are you able to provide the code you used to produce that python output? I don't see any code in the MCAP library itself that would produce a `compression.compression_ratio` value.

Hi，I write this python code refer to mcap’s cli code.😁😁

closed 2023-03-09 by HyokaChen