foxglove/mcap #851: capnp transcoder and support in mcap cat --json output

jon-chuang, 2023-03-13:
**Description**
We want to be able to decode capnp with dynamic schema in  `mcap-support` e.g. as is done for protobuf: https://github.com/foxglove/studio/blob/3767475a507097b3e4e046fc4d8b3a20135101f8/packages/mcap-support/src/parseChannel.ts#L125-L134

As a POC, we should first implement this in python and go using the dynamic schema interface using reflection via https://pkg.go.dev/capnproto.org/go/capnp/schemas

jon-chuang, 2023-03-13:
Looking at all the possible pathways, while it may be fun to explore WASM compilation (see e.g. https://medium.com/cyberbotics/porting-a-c-rendering-engine-to-webassembly-9c32d76c31f1) or using cgo to bind to the JSON transcoder (https://github.com/capnproto/capnproto/blob/master/c%2B%2B/src/capnp/compat/json.h), given the poor support of Capnproto in various languages, especially for browser, It does seem to make much more sense to encode the original capnp recording into a compatible format like JSON (or CBOR for compactness). Anything that decodes capnproto in a browser environment will not be zero-copy anyway.

defunctzombie, 2023-03-13:
@jon-chuang Please me mindful when using the bug label. The bug label is for broken functionality. If you believe this issue is a bug please explain what existing functionality that we've documented is not working, provide steps to reproduce, and describe what you expect to happen.